The-EG/OctoPrint-CameraSettings #134: [FR] Implement Controls for [imx708_wide]

Extra-Fox, 2023-03-19:
Raspberry Pi Camera V3 (Sony imx708_wide)

{
  "wide_dynamic_range": {
    "type": "bool",
    "default": "0",
    "value": "0",
    "flags": "modify"
  }
}

Possibly related to this issue? https://community.octoprint.org/t/pi-camera-v3-imx-chipset-based-cameras-not-working/49022/114

The-EG, 2023-03-19:
duplicate of #100 and already available in the latest release candidate version.

And yes, the newer imx chipset cameras require some extra setup to get working, but that's outside the scope of this plugin. It should still work, although I've heard some reports that the controls are split over multiple devices (ie. 'cameras').

closed 2023-03-19 by The-EG

Extra-Fox, 2023-03-19:
Okay, well the plugin prompted me to log an issue all the same. So maybe that needs to be removed?

Extra-Fox, 2023-03-19:
Nevermind, I see you said "release candidate". I can wait for the public version. Got enough problems getting the camera just to work in the first place.