The-EG/OctoPrint-CameraSettings #135: [BUG] Video stream: Colors are not displaing properly

xtomasnemec, 2023-03-20:
**Observed behavior**
Colors are not displaing properly
They are working properly when i use `sudo service webcamd restart`
[image]

**OctoPrint Version**
1.8.7

**CameraSettings Plugin Version**
0.4.1

**Camera**
Pi Camera V2.1

**Linux/OctoPi**
Octopi 1.0.0

**Other Plugins**
No

The-EG, 2023-03-20:
And this does not happen when not using the plugin?
The plugin doesn't affect the stream other than setting the specific settings, and this would normally be related to issues outside of this plugin.

xtomasnemec, 2023-03-20:
No @The-EG

The-EG, 2023-03-20:
What settings have you set? Can you narrow it down to specific settings? Again, this is normally indicative of an issue with the streamer or camera itself and not this plugin.

xtomasnemec, 2023-03-20:
I only changed the bitrate settings to: Constant Bitrate, Video bitrate: 25000000

xtomasnemec, 2023-03-20:
I think i could set `sudo service webcamd restart` to run at boot but i don't know how to do that.

The-EG, 2023-03-20:
Ah, so it does this *until* you restart `webcamd`?

xtomasnemec, 2023-03-20:
yes

xtomasnemec, 2023-03-20:
I added