The-EG/OctoPrint-CameraSettings #140: Error: No Focus change for Pi Cam 3 NoIR

Nomo1-7, 2023-03-23:
Pi Cam 3 NoIR

dw9807 10-000c manual focus
Main Problem: Focus not working. Default value 925, no focus change if changed from 0 to max

Nomo1-7, 2023-03-27:
[image]
[image]
[image]

The-EG, 2023-04-15:
Go to Logging in your OctoPrint settings, create a 'DEBUG' entry for this plugin, reproduce the above behavior again, and then attach your `octoprint.log` here, which can be found on the Logging settings page.

nomo17, 2023-05-04:
[octoprint-2.log](https://github.com/The-EG/OctoPrint-CameraSettings/files/11399759/octoprint-2.log)

fosterthree, 2023-05-21:
Having the same issue - moving the slider does not change the focus. [Log attached.](https://github.com/The-EG/OctoPrint-CameraSettings/files/11525284/octoprint.1.log)

The-EG, 2023-07-02:
Hey there, neither of those logs show you actually trying to change a setting.

The-EG, 2024-01-07:
Closing this for now since there's been no activity.

closed 2024-01-07 by The-EG