The-EG/OctoPrint-CameraSettings #143: Exposure Time Absolute adjustment is semi random

Brynyard, 2023-03-26:
**Observed behavior**
Tried to adjust ETA to a precise value to sync with PWM frequency of LED's to avoid flickering. Clicking up/down increase/decrease the value by a random value. The slider is so imprecise it's impossible to set any meaningful value.

**Expected behavior**
Click up/down should increase/decrease by 1 (or any other fixed value)

**OctoPrint Version**
1.8.7

**CameraSettings Plugin Version**
0.4.1

**Camera**
Logitech C505

**Linux/OctoPi**
OctoPi, it has versions?

**Other Plugins**
No

The-EG, 2023-03-26:
This is not a bug. The sliders move in steps that are reported as supported by your particular camera. Your browser may be overriding this. I have control of neither.

You can always enter a value directly into the box to the left of the slider instead, but again, there's no guarantee that the camera will support the exact value since many report range values with specific steps. It may pick something close, it may do nothing 🤷‍♂️ .

closed 2023-03-26 by The-EG

Brynyard, 2023-03-26:
Uhm, didn’t know that a random value between 0-20 would be classified as a «step», and if I enter a value it is set to <value> +/- 10. But if that os supposed to work.. erm.. welll, then I’d be really carefull using anything else in this plugin.

Brynyard, 2023-03-26:
To clarify: I can set it to any int value, and the camera also supports this (easy to test on cmdline), and the ETA slider is the only slider that has this «feature, definetly not a bug». It’s just seriously tedious having to set the value by retrying 5-6 times till the random value matches the expected value.