The-EG/OctoPrint-CameraSettings #157: [FR] Implement Controls for [Camera Name]

ebric47, 2023-04-17:
**Camera manufacturer and model Cam023 Raspberry IR-CUT Camera Automatic Version)**:

**Unknown controls information ({
  "hevc_sequence_parameter_set": {
    "type": "unknown"
  },
  "hevc_picture_parameter_set": {
    "type": "unknown"
  },
  "slice_param_array": {
    "type": "unknown"
  },
  "hevc_scaling_matrix": {
    "type": "unknown"
  },
  "hevc_decode_parameters": {
    "type": "unknown"
  }
})**:

The-EG, 2023-04-17:
duplicate of #106

closed 2023-04-17 by The-EG

ebric47, 2023-04-17:
Thanks, issue resolved Ed Brickey From: Taylor TalkingtonSent: Monday, April 17, 2023 1:38 PMTo: The-EG/OctoPrint-CameraSettingsCc: ebric47; AuthorSubject: Re: [The-EG/OctoPrint-CameraSettings] [FR] Implement Controls for [Camera Name] (Issue #157) duplicate of #106