The-EG/OctoPrint-CameraSettings #178: [FR] Implement Controls for [Camera Name]

doncounts, 2023-05-24:
**Camera manufacturer and model (ie. NexiGO N660)**:

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:{
  "hevc_sequence_parameter_set": {
    "type": "unknown"
  },
  "hevc_picture_parameter_set": {
    "type": "unknown"
  },
  "slice_param_array": {
    "type": "unknown"
  },
  "hevc_scaling_matrix": {
    "type": "unknown"
  },
  "hevc_decode_parameters": {
    "type": "unknown"
  }
}

doncounts, 2023-05-24:
The web cam was working until I upgraded to 1.9.0 I'm not sure what changed

The-EG, 2023-05-25:
duplicate of #106. This would not affect the webcam working, these are just controls that the newer kernel driver exposes.

closed 2023-05-25 by The-EG