The-EG/OctoPrint-CameraSettings #185: [FR] Implement Controls for [Camera Name]

GJSchaller, 2023-05-30:
Camera: Logitech C922

Controls Information:

{
  "hevc_sequence_parameter_set": {
    "type": "unknown"
  },
  "hevc_picture_parameter_set": {
    "type": "unknown"
  },
  "slice_param_array": {
    "type": "unknown"
  },
  "hevc_scaling_matrix": {
    "type": "unknown"
  },
  "hevc_decode_parameters": {
    "type": "unknown"
  }
}

The-EG, 2023-05-31:
Hey there, can you please verify that you are using the latest version of this plugin? 0.4.2

GJSchaller, 2023-05-31:
I should be - I just updated to OctoPrint 1.9.0 and updated all my plugins as of yesterday (May 30th).  I can check again when home tonight.

The-EG, 2023-05-31:
Thanks. I should have been fixed already, but if not I'll have to add in a few things to eliminate it for good. Also, it shouldn't be an issue in the mean time, it's not like you can do anything with these controls anyway.

GJSchaller, 2023-06-03:
I should note, C922 appears in the dropdown, below "rpivid" - the error comes up when I select rpivid.  It didn't occur to me until now that the error on rpivid may be extra / can be ignored, if the other choice is what I should be using.  (If I see an error, I tend to report it to be safe.  If I can ignore the error under rpivid, I apologize for the false alarm.)

The-EG, 2023-06-03:
Yes, you should be using the one for your logitech webcam. It does sound like this issue is still present though, so I'll have to fix it. Once it's fixed, the rpivid shouldn't show up at all since you can't really control much with it.

commit 5639a41a (2023-07-02): 🐛 Ignore unknown controls (fixes #185)

No really, I'm serious this time!
closed 2023-07-02 by The-EG