The-EG/OctoPrint-CameraSettings #197: [FR] Implement Controls for [Camera Name]

Loadmaster1, 2023-06-16:
**Camera manufacturer and model (ie. NexiGO N660)**:

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "hevc_sequence_parameter_set": {
    "type": "unknown"
  },
  "hevc_picture_parameter_set": {
    "type": "unknown"
  },
  "slice_param_array": {
    "type": "unknown"
  },
  "hevc_scaling_matrix": {
    "type": "unknown"
  },
  "hevc_decode_parameters": {
    "type": "unknown"
  }
}

Loadmaster1, 2023-06-16:
I have no idea if this is an issue or not.  I'm just following the instructions as shown.  Hope this helps.  Thanks!!

The-EG, 2023-06-17:
Thanks. not an issue, these aren't useful controls and can be ignored. They were already supposed to be ignored, and will be fixed in a future update.

closed 2023-06-17 by The-EG