The-EG/OctoPrint-CameraSettings #207: Camera settings are never saved.

paulbeard, 2023-12-31:
**Observed behavior**
Camera settings are never saved. The view is always 90° or more degrees out. Also the 50/60Hz choice is never saved. 

**Expected behavior**
When I hit the save button, those settings should be saved. 

**OctoPrint Version**
What version of OctoPrint are you using?
Version 1.9.3

**CameraSettings Plugin Version**
What version of this plugin are you using?
Camera Settings (0.4.3)

**Camera**
What camera are you using?
Doesn't seem to matter: Raspberry π module 2 and an Arducam 

**Linux/OctoPi**
What Linux distribution/image is this running on? If you are unsure, it is probably OctoPi. If OctoPi, what version?
Linux anycubic-pi 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux

**Other Plugins**
Do you have any other plugins that may be modifying the camera stream (ie, The Spaghetti Detective)?
Not that one. Mostly the default set.

The-EG, 2024-01-01:
Please enable debug logging for this plugin (Setings->Logging, create a new entry for camera settings and set it to debug), reproduce the behavior and then attach your octoprint.log here. And please include an entire section of the log, not just what you think may be relevant.

paulbeard, 2024-01-01:
I don't see where to create a new entry but I do see my camera view is once
again (after setting it 10 minutes ago) once again upside down and
backwards with 50Hz as the setting.

paulbeard, 2024-01-06:
Still no idea how to create a new entry. Still dealing with the camera setting not being saved.

The-EG, 2024-01-07:
Detailed instructions: https://github.com/The-EG/OctoPrint-CameraSettings/issues/34#issuecomment-840014617

Btw, impatience does not motivate me to do this any faster than I am currently able to *IN MY FREE TIME*.

closed 2024-01-07 by paulbeard