The-EG/OctoPrint-CameraSettings #211: ESP32 CAM ONVIF not work

Jumbo125, 2024-03-09:
Hy
thank you for your great work
When i start the octoprint server, i will get a fail, and i should post it here.
please can anybody help me please.
```
{
  "hevc_entry_point_offsets": {
    "type": "u32",
    "min": "0",
    "max": "4294967295",
    "step": "1",
    "default": "0"
  }
}
```

The-EG, 2024-03-10:
Hi, thanks for the submission. I'll work on getting this added so the warning goes away, but it won't keep you from using the plugin with your camera for now.

Jumbo125, 2024-03-26:
it doesn't work. 
here is a picture
[image]

[image]